SoonerRobotics/robomagellan-2019 #24: Create LIDAR class for data processing according to the Development Manual

jkleiber, 2018-11-27:
The LIDAR sensor has a pretty detailed datasheet posted in the google drive that outlines how to get data from it. We are interested in getting the distance to the nearest obstacle at 0.63 degree increments, and also knowing which direction is forward.

We want to be able to handle the data processing on the raspberry pi so this will end up being part of the SLAM/obstacle detection module on the main board.

For a starting point we need to be able to get the data from the device and output it to the console of the Pi.

jkleiber, 2018-11-27:
This likely will end up being a python script that will interface with the SLAM/obstacle detection program (which will also be written in python). I am going to write a python class for the data collection/telling the lidar what to do and another script that uses this class to read data about obstacles

colton5007, 2018-12-04:
http://www.ydlidar.com/download

closed 2019-01-09 by jkleiber